OpenCyphal-Garage/libcyphal #313: DSDL Identification number

kmilo7204, 2020-11-11:
Hi all,

I will try to explain my issue as good as I can (I hope it helps other people)

Currently I am using legacy-v0 (For Libuavcan and dsdl).

I have an issue related with the DSLD identification numbers (I will explain it better). I am able to use the standard and custom messages (Created with the dsdl_compiler) in different UAVCAN nodes to publish and subscribe to my data.

During my testing phase I found something that I am not able to understand (It is not clear for me) and is regarding my message ID. (Please see the image provided below) 
[image]

From this composed number it is easy for me to understand two things:
XX | Message-ID | Node-ID 

In this case I am publishing a message of type [uavcan/protocol/debug/KeyValue] in the node number 2; however, I do not understand where the first two numbers (18) are defined, it is not clear for me if I can modify that value. 

In addition it also seems weird for me that both of the publishers node starts with 10 instead and both of the published messages in both nodes start with  18  (I attach another image to provide a full view of what I just describe)

[image]

Thanks for your help!!

Please let me know if more information is required.

pavel-kirienko, 2020-11-11:
I think this link will answer your question: https://legacy.uavcan.org/Specification/4._CAN_bus_transport_layer/#can-frame-format (section "ID field")

kmilo7204, 2020-11-11:
Dear Pavel,

Thank you!!! 

Love this library!!

closed 2020-11-11 by kmilo7204