OpenCyphal-Garage/libcyphal #315: Cannot get SITL working

CorruptName, 2021-06-05:
So I've been trying for hours to get this simulator working. Not sure if it's just very terrible instructions or I'm missing something.
I'm on windows 10 and with Cygwin I get:
`$ ../Tools/autotest/sim_vehicle.py -v ArduPlane --console --osdmsp
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
`
So nothing. Not even an error message.
If I use WSL, I get much further:
`$ python ../Tools/autotest/sim_vehicle.py -v ArduPlane --console --osdmsp
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/mnt/c/Users/Corrupt/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl" "--osd"
Setting top to                           : /mnt/c/Users/Corrupt/ardupilot
Setting out to                           : /mnt/c/Users/Corrupt/ardupilot/build
Autoconfiguration                        : enabled
Setting board to                         : sitl
Using toolchain                          : native
Checking for 'g++' (C++ compiler)        : /usr/bin/g++
Checking for 'gcc' (C compiler)          : /usr/bin/gcc
Checking for c flags '-MMD'              : yes
Checking for cxx flags '-MMD'            : yes
CXX Compiler                             : g++ 7.5.0
Checking for need to link with librt     : not necessary
Checking for feenableexcept              : yes
Checking for HAVE_CMATH_ISFINITE         : yes
Checking for HAVE_CMATH_ISINF            : yes
Checking for HAVE_CMATH_ISNAN            : yes
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes
Checking for header endian.h                   : yes
Checking for header byteswap.h                 : yes
Checking for HAVE_MEMRCHR                      : yes
Checking for program 'python'                  : /usr/bin/python
Checking for python version >= 2.7.0           : 3.6.9
Checking for program 'python'                  : /usr/bin/python
Checking for python version >= 2.7.0           : 3.6.9
Source is git repository                       : yes
Update submodules                              : yes
Checking for program 'git'                     : /usr/bin/git
Checking for program 'size'                    : /usr/bin/size
Benchmarks                                     : disabled
Unit tests                                     : enabled
Scripting                                      : enabled
Scripting runtime checks                       : enabled
Debug build                                    : disabled
Coverage build                                 : disabled
Checking for program 'rsync'                   : /usr/bin/rsync
'configure' finished successfully (4.800s)
SIM_VEHICLE: Building
SIM_VEHICLE: "/mnt/c/Users/Corrupt/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arduplane"
Waf: Entering directory `/mnt/c/Users/Corrupt/ardupilot/build/sitl'
Embedding file locations.txt:Tools/autotest/locations.txt
Embedding file models/Callisto.json:Tools/autotest/models/Callisto.json
[5/7] Processing uavcangen: libraries/AP_UAVCAN/dsdl/ardupilot libraries/AP_UAVCAN/dsdl/com
Traceback (most recent call last):
  File "/mnt/c/Users/Corrupt/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc", line 59, in <module>
    from libuavcan_dsdl_compiler import run as dsdlc_run
  File "/mnt/c/Users/Corrupt/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 18, in <module>
    from uavcan import dsdl
ModuleNotFoundError: No module named 'uavcan'`

But if I "pip install uavcan" I get namespace errors.
Before you suggest that I'm missing dependencies, I have already done:
`git submodule update --init`
Any ideas?

pavel-kirienko, 2021-06-05:
Please re-open this question at https://github.com/ArduPilot/ardupilot/issues

closed 2021-06-05 by pavel-kirienko

CorruptName, 2021-06-05:
Sorry, Wrong tab LOL.