OpenCyphal-Garage/libcyphal #372: Remove `None` as a scheduling option

pavel-kirienko, 2024-08-06:
>This special case doesn't look convincing. I understand that there is some value in being able to unschedule a callback but as far as I understand by now we haven't identified an explicit need for that, and it would be easy to retrofit -- if needed -- later. The troubling thing here is that an extension of any well-structured set of entities with a special case often tends to break the design in unexpected ways at some point.
>
>>*But I couldn’t resist the temptation to put in a null reference, simply because it was so easy to implement.*

_Originally posted by @pavel-kirienko in https://github.com/OpenCyphal-Garage/libcyphal/pull/371#discussion_r1704753104_

serges147, 2024-08-07:
- The case of callback schedule cancellation could be covered by adding (in the future, if needed) by having new `callback.cancel()` method - callback will remain fully function but just in a dormant state, ready to be resheduled.
- `executor.scheduleCallback(callback, ...)` -> `callback.schedule(...)` move will eliminate another point of failure which is currently described in docs as "_...must be used only with the same executor..._"

serges147, 2024-08-14:
Closing as fixed by PR #376

closed 2024-08-14 by serges147