OpenCyphal-Garage/libcyphal #375: Add -Wno-error=attributes if the current compiler is GCC

pavel-kirienko, 2024-08-11:
Because older versions of GCC do not recognize `[[unused]]` and hence break the build.

serges147, 2024-08-12:
@pavel-kirienko Please give me details how to reproduce it. F.e. what exact "older" version of GCC I should try on? Thanks!
BTW, do we have somewhere documented which toolsets and their minimal versions we support?

pavel-kirienko, 2024-08-12:
I was trying to build it on my laptop last night which has GCC 11, which produced this warning. This is a known issue; in another codebase we have this:

```cmake
# Disable GCC warnings that produce false positives due to bugs in the compiler.
if (CMAKE_CXX_COMPILER_ID STREQUAL "GNU")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wno-unknown-attributes -Wno-maybe-uninitialized -Wno-uninitialized")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wno-error=stringop-overread")
endif ()
```

Clang appears to be much more stable btw.

closed 2025-02-04 by pavel-kirienko