OpenCyphal-Garage/libcyphal #404: registerCallback returning `Any` considered harmful

thirtytwobits, 2024-11-23:
In the current design all of the places where we have `Any registerCallback(Function&&)` inhibit the ability of users to author custom executors and to integrate with various platforms. The primary issue is the method is required to move a node into an opaque container that is then moved into an unknown caller. There is no canonical key available for the implementation of registerCallback to retain. This prohibits filtering callbacks within execution stages based on type and writing event-driven rescheduling of callbacks by platform layers.

We need to modify this design to use a allocation pool instead of Any types so there is only even a single memory location containing a given callback within a process.

pavel-kirienko, 2024-11-25:
I should add that the minimization of dynamic memory allocation (incl. pmr and pools) was one of the core design goals, hence why this design was chosen:

[image]

thirtytwobits, 2024-11-25:
If you look at #402 you'll see my attempts to attach custom properties to these nodes are thwarted by unknown and mysterious forces lurking in the depths of CAVL.

thirtytwobits, 2025-01-29:
I'm closing this issue in favor of another. As I'm working though an official FreeRTOS integration I'm learning more about the design. Mostly things are going smoothly and integrating with FreeRTOS+TCP proceeds apace, however, the single_threaded_executor is wholly unusable in FreeRTOS. This would be okay (i.e. I can just implement IExecutor) except we have a lot of important logic in there to support scheduling that I can't reuse without copy-and-paste.

Okay. Opening new issue and continuing this conversation.

closed 2025-01-29 by thirtytwobits