OpenCyphal-Garage/libcyphal #414: Size/Speed optimization: create O(n) version of chronomorphize

thirtytwobits, 2025-01-24:
The O(1) version of chronomorphize generates huge amounts of program text in binaries. Add a build option allowing users to enable this O(1) where there is more storage or use the O(n) version where storage is limited.

I suggest making the O(n) version the default since a linear-time search shouldn't be a surprising default but the amount of text generated by the O(1) optimization is surprising.

thirtytwobits, 2025-02-02:
Something is deeply wrong with chronomorphize. In this 44-frame callstack (which is a bit insane in and of itself) it takes 76761 characters to describe this one line. Furthermore, as optimization levels increase there is very little reduction in the callstack as one might expect when using template. At -03 (an optimization level one should never use for embedded) this remains a callstack with 33 levels of depth. 

cetl::pf17::detail::var::chronomorphize_impl<std::integer_sequence<unsigned int, 0u, 1u, 2u, 3u, 4u, 5u, 6u, 7u, 8u> >::lookup<cetl::pf17::detail::var::chronomorphize<cetl::pf17::detail::var::visit<cetl::overloaded<adn::freertos::applib::cyphal::udp::TransientErrorReporter<0u>::operator()(cetl::pf17::variant<libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxPublish, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRequest, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRespond, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxMsgReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxSvcReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeRxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeTxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaTxSocketSend, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaRxSocketReceive>&)::{lambda(libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxPublish const&)#1}, adn::freertos::applib::cyphal::udp::TransientErrorReporter<0u>::operator()(cetl::pf17::variant<libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxPublish, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRequest, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRespond, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxMsgReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxSvcReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeRxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeTxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaTxSocketSend, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaRxSocketReceive>&)::{lambda(libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRequest const&)#1}, adn::freertos::applib::cyphal::udp::TransientErrorReporter<0u>::operator()(cetl::pf17::variant<libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxPublish, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRequest, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRespond, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxMsgReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxSvcReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeRxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeTxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaTxSocketSend, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaRxSocketReceive>&)::{lambda(libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRespond const&)#1}, adn::freertos::applib::cyphal::udp::TransientErrorReporter<0u>::operator()(cetl::pf17::variant<libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxPublish, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRequest, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRespond, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxMsgReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxSvcReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeRxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeTxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaTxSocketSend, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaRxSocketReceive>&)::{lambda(libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxMsgReceive const&)#1}, adn::freertos::applib::cyphal::udp::TransientErrorReporter<0u>::operator()(cetl::pf17::variant<libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxPublish, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRequest, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardTxRespond, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxMsgReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxSvcReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeRxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeTxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaTxSocketSend, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaRxSocketReceive>&)::{lambda(libcyphal::transport::udp::IUdpTransport::TransientErrorReport::UdpardRxSvcReceive const&)#1}, adn::freertos::applib::cyphal::udp::TransientErrorReporter<0u>::operator()(cetl::pf17::variant<libcyphal::transport::udp::IUdpTransport::Tra [... cut ...] t::udp::IUdpTransport::TransientErrorReport::UdpardRxSvcReceive, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeRxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaMakeTxSocket, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaTxSocketSend, libcyphal::transport::udp::IUdpTransport::TransientErrorReport::MediaRxSocketReceive>, 2> const&)::{lambda(auto:1)#1}&&) const@0x0046d486 (external/cetl/include/cetl/pf17/variant.hpp:91)

pavel-kirienko, 2025-02-04:
Moved https://github.com/OpenCyphal/CETL/issues/157

closed 2025-02-04 by pavel-kirienko