OpenCyphal-Garage/libcyphal #416: Impossible to maintain transfer ids.

serges147, 2025-01-28:
Please see corresponding topic at PyCyphal : https://pycyphal.readthedocs.io/en/stable/api/pycyphal.presentation.html#pycyphal.presentation.Presentation.output_transfer_id_map

Problem was found during OCVSMD development, where multiple, quick (1Hz) in succession runs of SDK CLI tool fail to deliver "exec cmd" invocations. The use case and the reason of the failure is well described in the above link.

Todo:
- [ ] Introduce at the `presentation` layer a new `ITransferIdMap` interface. Extend `Presentation` class with `setTransferIdMap` method.
- [ ] Implement default (out of box) realization (`DefaultTransferIdMap` ? TBD)
    Will do nothing, always return zeros for a new publisher or rpc client. Will be used as default fallback to the current behavior. It will be up to libcyphal client to setup his own better suited implementation.
- [ ] Use the interface to get/set transfer ids at presentation ...
  - [ ] `PublisherImpl`
  - [ ] `SharedClient`
- [ ] Mock and unit tests
- [ ] Try at OCVSMD:
  - [ ] Implement `ITransferIdMap` with map in memory, and persisting to file (on exit).
  - [ ] Demo the use case.

thirtytwobits, 2025-01-28:
This seems well-aligned with pycyphal. Seems like a reasonable design.

serges147, 2025-01-31:
Fixed by pr #420