OpenCyphal-Garage/libcyphal #426: IRxSocket media data buffer type should be opaque

thirtytwobits, 2025-02-01:
We currently use `std::unique_ptr<cetl::byte, PmrRawBytesDeleter> payload_ptr` which forces me to use a `PmrRawBytesDeleter` which is incorrect for FreeRTOS zero copy. Instead I need to have a deleter that invokes `FreeRTOS_ReleaseUDPPayloadBuffer`.

One option is to set the deleter type in libcyphal/config.hpp (see #419). This doesn't support heterogenous socket implementations however. Other ideas are welcome. We should also plan for different media layers that may need an additional data pointer as well; one for the payload data and another that is the key for the buffer in a memory pool (where the two are not the same thing. I'd tell you how FreeRTOS+TCP avoids the need for this additional data but you really don't want to know).

pavel-kirienko, 2025-02-04:
The idea is that you will have your own pmr that invokes `FreeRTOS_ReleaseUDPPayloadBuffer`. Am I missing something in the problem formulation?

pavel-kirienko, 2025-02-04:
See https://github.com/OpenCyphal-Garage/libcyphal/issues/352#issuecomment-2155653605

serges147, 2025-04-22:
> The idea is that you will have your own pmr that invokes `FreeRTOS_ReleaseUDPPayloadBuffer`. Am I missing something in the problem formulation?

exactly!