OpenCyphal-Garage/libcyphal #429: TransportImpl::peekFirstValidTxItem has unbounded loop

thirtytwobits, 2025-02-02:
While loop in this method has no time-bound exit criteria. A large number of expired frames will cause this method to consume a proportionally large amount of CPU time.

thirtytwobits, 2025-02-02:
Also found the same condition in `sendNextFrameToMediaTxSocket`

pavel-kirienko, 2025-02-04:
Our scheduler has no notion of execution time limits, but we could add a fixed number of frames to traverse before cutting the call short. This should be enough for MISRA/AUTOSAR compliance.

serges147, 2025-04-21:
>A large number of expired frames will cause this method to consume a proportionally large amount of CPU time.

@thirtytwobits 
I don't think it's correct assessment. Expired items will be dropped immediately (by `popAndFreeCanardTxQueueItem`), so they won't stay there, and next call to `peekFirstValidTxItem` won't encounter them again.

Could you please describe concrete scenario in which we might encounter "a large number of expired frames"?
Especially for CAN transport where we implemented proactive purging of expired items during push (see `::canardTxPoll` and its `txFlushExpiredTransfers` call).