OpenCyphal-Garage/libcyphal #431: duration cast missing when using 'ard µsec

thirtytwobits, 2025-02-02:
We need to search for all places in the code that do this:
```
 TimePoint{std::chrono::microseconds{out_transfer.timestamp_usec}};
```
and replace them with this:
```
 TimePoint{std::chrono::duration_cast<Duration>(std::chrono::microseconds{out_transfer.timestamp_usec})};
```

serges147, 2025-03-03:
@thirtytwobits @pavel-kirienko Please elaborate - I don't see what is the issue here. I believe that constructor # 4 does exactly that: https://en.cppreference.com/w/cpp/chrono/duration/duration

serges147, 2025-03-03:
ok now I see it - constr#4 does it only if:
>This overload participates in overload resolution only if no overflow is induced in the conversion, and any of the following conditions is satisfied:...

So, it will compile fine if I change our `TimePoint::period` to nanoseconds (b/c 10^9 divisible by 10^6 without reminder), BUT it fails to compile if I try use milliseconds (b/c (`10^3 % 10^6) != 0`) - this is the case when you have to use the explicit `duration_cast`, kinda being ok with potential precision loss.
@thirtytwobits Could you please confirm, that this is exactly (as I suppose you tried milliseconds) what you have encountered?

serges147, 2025-04-22:
Fixed by pr #441

closed 2025-04-23 by serges147