OpenCyphal-Garage/libcyphal #432: UDP Media Layer send method's design is incorrect

thirtytwobits, 2025-02-02:
The current media layer abstraction calls "send" on a socket with a span of spans to transmit. For embedded systems this is difficult to handle as the platform's send method may return an out-of-resources error which cannot be cleared until/unless the callstack unwinds and a yield point is reached to allow the scheduler to give the network driver CPU time.

While we currently only ever send a single span in this span-of-spans, the interface doesn't let the developer know about this behavioural contract. The implementer is left to try to handle the type of complex timing and resource management issues libcyphal is meant to alleviate.

We should simply change this to always have a single fragment in the method.

pavel-kirienko, 2025-02-04:
>The current media layer abstraction calls "send" on a socket with a span of spans to transmit. For embedded systems this is difficult to handle as the platform's send method may return an out-of-resources error which cannot be cleared until/unless the callstack unwinds and a yield point is reached to allow the scheduler run the network driver.

How is this different for the case when there is only one span? All spans belong to the same UDP datagram / CAN frame / whatever else transport-specific entity there is, so they either go out together, or they are dropped together.

pavel-kirienko, 2025-04-09:
Can we close this?

closed 2025-12-17 by pavel-kirienko