OpenCyphal-Garage/libcyphal #434: Missing way to return failure from registerCallback method signatures

thirtytwobits, 2025-02-03:
In places where we call `IExecutor::Callback::Any registerCallback(IExecutor::Callback::Function&& function)` on object (e.g. IRxSocket in the udp transport) there is no well-defined way to return a failure.

pavel-kirienko, 2025-02-03:
It is designed to be infallible. In particular, implementations are not expected to allocate memory. What failure modes do you think are missing?

thirtytwobits, 2025-02-03:
I'm using flyweight objects which are not real sockets. There is a chance that the real socket got deleted which leaves the flyweights as orphans. I have no way to signal that the object is no longer initialized even if the C++ object remains well-formed.

Additionally, the only way I can schedule sockets after the `registerCallback` method was called is to save a pointer to the callback's interface. `getInterface` can return null and I have no way to signal that this isn't acceptable; that the callback created by the IExecutor is not valid for the socket.

pavel-kirienko, 2025-02-04:
Let us please look at your code